lvgl/lv_port_stm32f746_disco #4: Build error: junk at end of line

roy-zahor, 2021-02-14:
Hi,

Trying to compile the project using STM32CubeIDE v1.5.1, I'm getting the following error.

Can you please help?

```
11:41:51 **** Build of configuration Debug for project lv_port_stm32f746_disco ****
make all 

... ... ...

arm-none-eabi-gcc -mcpu=cortex-m7 -g3 -c -x assembler-with-cpp -MMD -MP -MF"ide/iar/startup_stm32f746xx.d" -MT"ide/iar/startup_stm32f746xx.o" --specs=nano.specs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -o "ide/iar/startup_stm32f746xx.o" "../ide/iar/startup_stm32f746xx.s"
../ide/iar/startup_stm32f746xx.s: Assembler messages:
../ide/iar/startup_stm32f746xx.s:1: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:2: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:3: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:4: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:5: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:6: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:7: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:8: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:9: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:10: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:11: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:12: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:13: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:14: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:15: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:16: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:17: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:18: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:19: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:20: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:21: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:22: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:23: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:24: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:25: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:26: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:27: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:28: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:29: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:30: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:31: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:32: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:33: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:34: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:35: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:35: Error: bad instruction `loss OF USE,DATA,OR PROFITS'
../ide/iar/startup_stm32f746xx.s:35: Error: bad instruction `or BUSINESS INTERRUPTION)HOWEVER'
../ide/iar/startup_stm32f746xx.s:36: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:37: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:38: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:39: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:40: Error: junk at end of line, first unrecognized character is `*'
../ide/iar/startup_stm32f746xx.s:43: Error: bad instruction `the modules in this file are included in the libraries,and may be replaced'
../ide/iar/startup_stm32f746xx.s:44: Error: bad instruction `by any user-defined modules that define the PUBLIC symbol _program_start or'
../ide/iar/startup_stm32f746xx.s:45: Error: bad instruction `a user defined start symbol.'
../ide/iar/startup_stm32f746xx.s:46: Error: bad instruction `to override the cstartup defined in the library,simply add your modified'
../ide/iar/startup_stm32f746xx.s:47: Error: bad instruction `version to the workbench project.'
../ide/iar/startup_stm32f746xx.s:49: Error: bad instruction `the vector table is normally located at address 0.'
../ide/iar/startup_stm32f746xx.s:50: Error: bad instruction `when debugging in RAM,it can be located in RAM,aligned to at least 2^6.'
../ide/iar/startup_stm32f746xx.s:51: Error: bad instruction `the name "__vector_table"has special meaning for C-SPY:'
../ide/iar/startup_stm32f746xx.s:52: Error: condition required -- `it is where the SP start value is found,and the NVIC vector'
../ide/iar/startup_stm32f746xx.s:53: Error: bad instruction `table register(VTOR)is initialized to this address if!=0.'
../ide/iar/startup_stm32f746xx.s:55: Error: bad instruction `cortex-M version'
../ide/iar/startup_stm32f746xx.s:58: Error: bad instruction `module ?cstartup'
../ide/iar/startup_stm32f746xx.s:60: Error: bad instruction `forward declaration of sections.'
../ide/iar/startup_stm32f746xx.s:61: Error: bad instruction `section CSTACK:DATA:NOROOT(3)'
[... 539 lines cut ...]
../ide/iar/startup_stm32f746xx.s:702: Error: bad instruction `pubweak QUADSPI_IRQHandler'
../ide/iar/startup_stm32f746xx.s:703: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:704: Error: bad instruction `quadspi_irqhandler'
../ide/iar/startup_stm32f746xx.s:707: Error: bad instruction `pubweak LPTIM1_IRQHandler'
../ide/iar/startup_stm32f746xx.s:708: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:709: Error: bad instruction `lptim1_irqhandler'
../ide/iar/startup_stm32f746xx.s:712: Error: bad instruction `pubweak CEC_IRQHandler'
../ide/iar/startup_stm32f746xx.s:713: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:714: Error: bad instruction `cec_irqhandler'
../ide/iar/startup_stm32f746xx.s:717: Error: bad instruction `pubweak I2C4_EV_IRQHandler'
../ide/iar/startup_stm32f746xx.s:718: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:719: Error: bad instruction `i2c4_ev_irqhandler'
../ide/iar/startup_stm32f746xx.s:722: Error: bad instruction `pubweak I2C4_ER_IRQHandler'
../ide/iar/startup_stm32f746xx.s:723: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:724: Error: bad instruction `i2c4_er_irqhandler'
../ide/iar/startup_stm32f746xx.s:727: Error: bad instruction `pubweak SPDIF_RX_IRQHandler'
../ide/iar/startup_stm32f746xx.s:728: Error: bad instruction `section .text:CODE:NOROOT:REORDER(1)'
../ide/iar/startup_stm32f746xx.s:729: Error: bad instruction `spdif_rx_irqhandler'
../ide/iar/startup_stm32f746xx.s:731: Error: bad instruction `end'
make: *** [ide/iar/subdir.mk:18: ide/iar/startup_stm32f746xx.o] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

11:43:07 Build Failed. 615 errors, 19 warnings. (took 1m:15s.803ms)

```

roy-zahor, 2021-02-14:
Fixed by removing /ide/iar/startup_stm32f746xx.s.

embeddedt, 2021-02-14:
I need to adjust the Eclipse project to ignore that file.

embeddedt, 2021-03-06:
This should be fixed now. Thanks for the report!

commit 1162b479 (2021-03-06): Fix #4: ignore IAR and CMake files
closed 2021-03-06 by embeddedt