lvgl/lv_port_stm32f746_disco #6: build "lv_demo_printer()" fail

qq0901001, 2021-03-15:
c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: lv_port_stm32f746_disco.elf section `.rodata' will not fit in region `FLASH'
c:\st\stm32cubeide_1.6.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.9-2020-q2-update.win32_1.5.0.202011040924\tools\arm-none-eabi\bin\ld.exe: region `FLASH' overflowed by 192296 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:140: lv_port_stm32f746_disco.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

kisvegabor, 2021-03-15:
The printer demo uses a lot of images. If you use 32 bit color depth, try with 16 bit.

embeddedt, 2021-03-15:
Please note that last time I tested it, the printer demo is designed for the larger 800x480 screen found on STM32F769, and doesn't render properly on F746's 480x272 screen.

qq0901001, 2021-03-16:
@kisvegabor @embeddedt Thank you for you replay. ^_*

I haven't realized it is the resolution problem, so it's a pity.

LVGL Demo GUI on STM32F769 Discovery (800 x 480) exactually have a good performance.
[https://www.youtube.com/watch?v=I5iTb9AkY2w](url)
[https://www.youtube.com/watch?v=ZUJxbCbBod0](url)

kisvegabor, 2021-04-13:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2021-04-13 by kisvegabor