lvgl/lv_port_stm32f746_disco #18: 'missing and no known rule to make it' in IAR

GorgonMeducer, 2022-01-10:
I followed the instruction and clone (incursively) the repository. 
In IAR, I got the following error:

[image]

It appears that the path is not right for files under asserts:

[image]

[image]

I am using the latest IAR (9.10.2)

Can you please fix this issue? Thanks.

embeddedt, 2022-01-10:
I can't fix it, unfortunately, as I don't have IAR installed (and I use Linux, so I have no way of getting it easily).

I'll CC @canhkha as they originally contributed the project files.

GorgonMeducer, 2022-01-10:
After further checking, it appears that this IAR project is completely out of date and hasn't been updated for quite a long time.

embeddedt, 2022-01-10:
It might be more prudent to remove it for now, and it can be re-added if someone has interest in maintaining it. @kisvegabor What do you think?

kisvegabor, 2022-01-15:
Good question. Constantly removing unmaintained parts is also a burden. Besides, if we remove them people wont't see them and it's less likely that someone sends an update once. 

I general, in case of something becomes out dated, I think the best would be to mention that in the README and open a `good-first-issue` issue about it.  What do you think? We can even have an issue in the lvgl repo where we reference these issues.

embeddedt, 2022-01-21:
I'm concerned that leaving an unmaintained component in-tree might give the false impression that it still works (when it doesn't), but I don't have a better suggestion, so let's do it this way.

kisvegabor, 2022-01-21:
Okay, I update the README (cc9296c) and opened an issue (#19)

embeddedt, 2022-01-22:
Thanks! In that case, I suppose we can close this.

closed 2022-01-22 by embeddedt

abdalrahman01, 2023-01-01:
I had the same issue, I fixed it by updating the configurations: 
Project -> Edit Configurations  -> new (make a new one based on a previus ones)